NMminer1024/NMMiner #8: Mqtt possible

Bluelightcrew, 2024-11-08:
Is it possible to integrate an mqtt interface so that you can, for example, access the data in the smart home

NMminer1024, 2024-11-08:
> Yes, but mqtt will consume a certain amount of CPU resources, so the hashrate will be affected to a certain extent and decrease.
> So，what's the origin demands to connect to smart home system?

Yes, but mqtt will consume a certain amount of CPU resources, so the hashrate will be affected to a certain extent and decrease.
So，what's the origin demands to connect to smart home system?

Bluelightcrew, 2024-11-08:
Ok I understand, hashrate is more important. Thanks for the answer. I will se if I can pars it from the server page

Bluelightcrew, 2024-11-08:
Is it possible to flash it on a ESP32-2432S028R? I like this Firmware more with this web monitor!

NMminer1024, 2024-11-09:
> Is it possible to flash it on a ESP32-2432S028R? I like this Firmware more with this web monitor!

Sure ，https://flash.nmminer.com/  here you can flash it by yourself Sir.
[image]

NMminer1024, 2024-11-09:
> Ok I understand, hashrate is more important. Thanks for the answer. I will se if I can pars it from the server page

https://github.com/NMminer1024/NMController_web
Here is a reference for Web monitor ，hope that's a useful example for you.

closed 2024-11-16 by NMminer1024