NMminer1024/NMMiner #11: False "Block found"?

dwightmulcahy, 2025-01-08:
[image]

I updated to 5.0.2 the other day and noticed this today.  I immediately checked my wallet for any transactions; sadly, there were none.

dWiGhT

greghub-alt, 2025-01-08:
> ![PXL_20250108_224022355~2](https://private-user-images.githubusercontent.com/8507802/401331129-dd900014-edf9-4ee0-ad45-f0fcf8c36a87.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzNzY2OTEsIm5iZiI6MTczNjM3NjM5MSwicGF0aCI6Ii84NTA3ODAyLzQwMTMzMTEyOS1kZDkwMDAxNC1lZGY5LTRlZTAtYWQ0NS1mMGZjZjhjMzZhODcuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDhUMjI0NjMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTM5ZjNkZjI0NTQ0Yzg5OWU4MzU3ZjhmYWViNDA1NTFmNmM0ZjY2MTY3ODMwMGRhYjNmMDExMWM2M2I4ODdlZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.aPBapFXK5GNtOW9YyA4T-6XLguDahd-9vy9DgVuw7as)
> 
> I updated to 5.0.2 the other day and noticed this today. I immediately checked my wallet for any transactions; sadly, there were none.
> 
> dWiGhT

Hi there,

We use GitHub issues as a place to track bugs and other development-related issues. 

Please see the link below to our dedicated support line:

[Help Center :  Click Here](https://livesupports.app/github)

Ticket ID: WB240

Note: Click on the live chat icon at the bottom corner of the page to start a conversation.

NMminer1024, 2025-01-09:
> ![PXL_20250108_224022355~2](https://private-user-images.githubusercontent.com/8507802/401331129-dd900014-edf9-4ee0-ad45-f0fcf8c36a87.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY2 [...]

Running over 42 days，a reboot action should be done，I think it's a worng state，long press ‘boot’ button to clear all data storage

dwightmulcahy, 2025-01-09:
> Running over 42 days，a reboot action should be done，I think it's a worng state，long press ‘boot’ button to clear all data storage

The 42 days are from "saved stats"... I just flashed from 5.0.1 to 5.0.2 a couple of days ago.  Will give it a poke to clear it.

dWiGhT

closed 2025-01-12 by NMminer1024

klipklipklip, 2025-01-23:
I flashed my device with version 0.5.0.2 on December 29, 2024, and I’ve also been experiencing a false block since this morning.

dwightmulcahy, 2025-01-23:
I recently flashed all mine to v5.0.3, so far no false blocks indicated.