NMminer1024/NMMiner #18: Pool Selection / Algo Selection

Mecanik, 2025-03-11:
Hi,

Amazing project, congrats!

Bought this model for testing:

https://www.aliexpress.com/item/1005008484663300.html

However, I did not use public-pool.io because I cannot see it connect in my browser, hence I cannot verify the miner. So I used [btc.solopool.org](https://btc.solopool.org/), but I don't get any shares, difficulty, or anything, please see photo.

[image]

Additionally, does this chip support only BTC? And only solo?

Many thanks!

betterpony, 2025-03-11:
> 日　期：2025年03月11日 19:46:02
> 主　题：[NMminer1024/NMMiner] Pool Selection / Algo Selection (Issue #18)

show me a log from web monitor, I think you'd better set stratum password filed as 'd=0.001' which is suitable for NMMiner, that means let pool know you want the initial diff is 0.001, hope that's a helpful solution for your question

来自阿里邮箱Android客户端------------------------------------------------------------------

https://www.aliexpress.com/item/1005008484663300.html
1741693380764.jpg (view on web)
https://www.aliexpress.com/item/1005008484663300.html
1741693380764.jpg (view on web)

NMminer1024, 2025-03-11:
> Hi,
>
> Amazing project, congrats!
>
> Bought this model for testing:
>
> https://www.aliexpress.com/item/1005008484663300.html
>
> However, I did not use public-pool.io because I cannot see it connect in my [...]

Hi sir,
      Do you mind show me a log from web monitor, I think you'd better set stratum password filed as 'd=0.001' which is suitable for NMMiner, that means let pool know you want the initial diff is 0.001, hope that's a helpful solution for your question

Mecanik, 2025-03-11:
Thanks for replying. Here is what you asked:

[image]

1. I cannot update the firmware to latest 1.5 because on the web tool there are 2 versions for this model, and I'm not sure which one is it (esp32-2432s028r-ili9341 or esp32-2432s028r-st7789)
2. I am now testing ViaBTC, PPS+. I did not think about the difficulty, so I must find pool with very low diff correct?
3. Since this algorithm is SHA-256, possible mine BCH? (does the miner accept the wallet?)

Many thanks!!!

NMminer1024, 2025-03-11:
> Thanks for replying. Here is what you asked:
>
> [image]
>
> 1. I cannot update the firmware to latest 1.5 because on the web tool there are 2 versions for this model, and I'm not sure which one is it [...]

1. Never mind this，st7789 and ili9341 is the screen driver , so you could flash 7789 first，if you notice the screen color is weird，then flash ili9341，this 2 firmware is same except the driver
2. Yes，you have to chose a low difficulty pool first，diff around 0.0001~ 0.01 maybe suitable，but I think it's nothing reward at this kind of hashrate level.
3. Sure，miner accept any wallet address you put into if pool accept it，all sha256d algorithm support，such as https://pool.bchsolo.com/#bch/connect，pool.bchsolo.com:3332，d=0.001

NMminer1024, 2025-03-11:
> > > Thanks for replying. Here is what you asked:
> > > [image]
> > > 
> > > 1. I cannot update the firmware to latest 1.5 because on the web tool there are 2 versions for this model, and I'm not sure which one is it (esp32-2432s028r-ili9341 or esp32-2432s028r-st7789)
> > > 2. I am now testing ViaBTC, PPS+. I did not think about the difficulty, so I must find pool with very low diff correct?
> > > 3. Since this algorithm is SHA-256, possible mine BCH? (does the miner accept the wallet?)
> > > 
> > > Many thanks!!!
> > 
> > 
> > 
> > 1. Never mind this，st7789 and ili9341 is the screen driver , so you could flash 7789 first，if you notice the screen color is weird，then flash ili9341，this 2 firmware is same except the driver
> > 2. Yes，you have to chose a low difficulty pool first，diff around 0.0001~ 0.01 maybe suitable，but I think it's nothing reward at this kind of hashrate level.
> > 3. Sure，miner accept any wallet address you put into if pool accept it，all sha256d algorithm support，such as [https://pool.bchsolo.com/#bch/connect，pool.bchsolo.com:3332，d=0.001](https://pool.bchsolo.com/#bch/connect%EF%BC%8Cpool.bchsolo.com:3332%EF%BC%8Cd=0.001)
> 
> Thank you, I have successfully flashed to the latest firmware. The controller app is just life saving! If you need any help with it let me know, happy to pitch in!
> 
> Unfortunately I cannot mine anything else because no pool accepts such low difficulty, so I will order your other version for that (axe).
> 
> Last question please, how trustworthy is are these pools? (public-pool.io, pool.tazmining.ch) Asking because what if by a miracle a block is found... they will pay out no? I cannot find much about these pools.
> 
> Thanks again!

Regarding the last question, you can only choose to trust these mining pools, even though they have not hit a block so far, and there is no way to verify whether they will pay you after hitting a block. Once it is verified that they do not pay, I believe that the mining pool will also lose the trust of all miners.

Whatever, you should treat all this as a pleasant game, and don't pin all your hopes of life on hitting a block. Just enjoy it.

Mecanik, 2025-03-11:
> Regarding the last question, you can only choose to trust these mining pools, even though they have not hit a block so far, and there is no way to verify whether they will pay you after hitting a [...]

Of course, I fully understand this is pure lottery. But one day a block will drop to someone, and we want to ensure that person actually gets it, otherwise they will blame the miner's recommendation. I'm thinking of hosting a pool myself, a fully transparent and trustworthy one, but I cannot find any ready to use pool code.

NMminer1024, 2025-03-12:
> Of course, I fully understand this is pure lottery. But one day a block will drop to someone, and we want to ensure that person actually gets it, otherwise they will blame the miner's [...]

Here is a reference ： https://github.com/NMminer1024/btc-ckpool-solo

Mecanik, 2025-03-12:
Thanks, I will check it out. Meanwhile...

1. Can you please add this to the docs so people know... https://web.public-pool.io/
2. https://cointelegraph.com/news/solo-bitcoin-miner-wins-block-using-tiny-cheap-bitcoin-miner (prepare for mass orders LOL)

NMminer1024, 2025-03-12:
> Thanks, I will check it out. Meanwhile...
> 
> 1. Can you please add this to the docs so people know... https://web.public-pool.io/
> 2. https://cointelegraph.com/news/solo-bitcoin-miner-wins-block-using-tiny-cheap-bitcoin-miner (prepare for mass orders LOL)

Sure ，Thank you for your suggestion.