NMminer1024/NMMiner #19: LED Flashlight on ESP32 CAM Board

mathesgt, 2025-03-13:
Hi and first, thx for the work on this project 👍

I'm running the Firmware on an ESP32-CAM Board (ESP32 D0WD) that has a LED Flash LED on it. Sadly, the LED logic is reversed, so i'm not able to turn the very bright LED Flashing off with the LedEnable config switch, instead i turned it constantly on. Is it possible to add another config item for setting reverse logic on that function or change the boolean to values like "on", "blink", "off" - that would be really helpful :)

NMminer1024, 2025-03-13:
> Hi and first, thx for the work on this project 👍
>
> I'm running the Firmware on an ESP32-CAM Board (ESP32 D0WD) that has a LED Flash LED on it. Sadly, the LED logic is reversed, so i'm not able to [...]

Sure sir

mathesgt, 2025-03-19:
Hi,
big thanks! worked 24 hours now wothout any issues.

closed 2025-03-19 by mathesgt