NMminer1024/NMMiner #22: Not getting passed 86%

HarukiToreda, 2025-03-18:
I got a new ESP32-2432S028 and loaded it up with v1.6.01, got my license and all. But for some reason it is stuck loading at 86%

[image]

[I/NM] WiFi connected!
[I/NM] (market) thread started on core 0...
[I/NM] Market Connected
[I/NM] (swarm) thread started on core 0...
[I/NM] (stratum) thread started on core 0...
[W/NM] Resolving pool address [public-pool.io] to IP [38.51.144.240]
ublic-pool.io:21496]
[I/NM] Sending mining.subscribe : {"id": 1, "method": "mining.subscribe", "params": ["NMMiner"]}

rize : {"id": 2, "method": "mining.authorize", "params": ["de747763521230052578714d7dbd89b2404c04229a0c5a8526c1eb19a6bbb6df194b31a89e913da8x", "x"]}

[I/NM] (monitor) thread started on core 0...
[W/NM] (tMiner) mode 0xaa started on core 1...
[W/NM] (nMiner) mode 0xbb started on core 0...
[L/NM] Pool difficulty set : 0.00010 
[W/NM]  ++++++++++++++++++++++++++++++++++++++++++++++ Swarm +++++++++++++++++++++++++++++++++++++++++++++++++++++++
[L/NM] |          Best Diff           |               Workers             |               Total HashRate            |
[L/NM] |           176.0              |                  5                |                  1.252MH/s              |
[L/NM] Hello pool...
[L/NM] Hello pool...
[W/NM]  ++++++++++++++++++++++++++++++++++++++++++++++ Swarm +++++++++++++++++++++++++++++++++++++++++++++++++++++++
[L/NM] |          Best Diff           |               Workers             |               Total HashRate            |
[L/NM] |           176.0              |                  5                |                  1.252MH/s              |

NMminer1024, 2025-03-19:
> I got a new ESP32-2432S028 and loaded it up with v1.6.01, got my license and all. But for some reason it is stuck loading at 86%
>
> [image]
>
> [I/NM] WiFi connected! [I/NM] (market) thread started on [...]

Hi sir,
   I  notice that you put a wrong wallet address in the filed as "de747763521230052578714d7dbd89b2404c04229a0c5a8526c1eb19a6bbb6df194b31a89e913da8x"，can not accept by pool ，not firmware issue.
    Acceptable wallet address start with "bc" or “1” ，please double check .

HarukiToreda, 2025-03-20:
you are right, that was the issue. Thank you

closed 2025-03-20 by HarukiToreda