NMminer1024/NMMiner #23: Won't get past market connect.

Grantrocks, 2025-03-20:
I have a valid license and everything however I am unable to get this to work. It establishes a valid connection to my network but won't go past that.
It just throws ssl error and end of file errors.

[image]

Also after a bit it starts throwing other errors.

[image]

NMminer1024, 2025-03-20:
Hi sir 
    1. Which board 
    2. Try to disable the “market price” feature in the config page and try it again
    3.Would you mind show me the details of your config page

NMminer1024, 2025-03-20:
> I have a valid license and everything however I am unable to get this to work. It establishes a valid connection to my network but won't go past that. It just throws ssl error and end of file errors.
> 
> [image]
> 
> Also after a bit it starts throwing other errors.
> 
> [image]

Hi sir
1. Which board
2. Try to disable the “market price” feature in the config page and try it again
3.Would you mind show me the details of your config page

Grantrocks, 2025-03-20:
It's a lilygo tdisplay s3.
Disabling the market feature didn't do anything. 
I excluded wifi info for privacy.

[image]

NMminer1024, 2025-03-21:
> It's a lilygo tdisplay s3. Disabling the market feature didn't do anything. I excluded wifi info for privacy.
> 
> [image]

Hi sir 
    What's the log now

Grantrocks, 2025-03-21:
Same as before but no market connect errors.