NMminer1024/NMMiner #24: Hashrate fall back to ~408 KH/s

oscar-rr, 2025-03-23:
Hi,

after some time all my ESP32-2432S028R fall back to ~408 KH/s

> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  461.24KH/s |    1/149   |  112.1T  |  0.0032   |  0.0112   |  0.0790   |   0   |   0.092% |  0'C | -44dBm |
> [W/NM] Market Disconnected
> [W/NM]  -------------------------------------000d 00:10:40--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  462.48KH/s |    1/149   |  112.1T  |  0.0032   |  0.0112   |  0.0790   |   0   |   0.116% |  0'C | -43dBm |
> [W/NM]  -------------------------------------000d 00:10:42--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  462.00KH/s |    1/149   |  112.1T  |  0.0032   |  0.0112   |  0.0790   |   0   |   0.139% |  0'C | -41dBm |
> [L/NM] #151 share accepted, 311ms
> [W/NM]  -------------------------------------000d 00:10:44--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  460.10KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.161% |  0'C | -41dBm |
> [W/NM]  -------------------------------------000d 00:10:46--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  442.91KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.185% |  0'C | -40dBm |
> [W/NM] Market Disconnected
> [W/NM]  -------------------------------------000d 00:10:48--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  462.32KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.208% |  0'C | -46dBm |
> [W/NM]  -------------------------------------000d 00:10:50--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  436.04KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.230% |  0'C | -47dBm |
> [W/NM]  -------------------------------------000d 00:10:52--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  408.46KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.251% |  0'C | -42dBm |
> [W/NM]  -------------------------------------000d 00:10:54--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  408.55KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.271% |  0'C | -47dBm |
> [W/NM]  -------------------------------------000d 00:10:56--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  408.59KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.291% |  0'C | -45dBm |
> [W/NM]  -------------------------------------000d 00:10:58--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  408.60KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.312% |  0'C | -41dBm |
> [W/NM]  -------------------------------------000d 00:11:00--------------v1.6.01-------------------------------------
> [W/NM] |  hash rate  | share(R/A) | net diff | pool diff | last diff | best diff | valid | progress | temp |  rssi  |
> [I/NM] |  408.59KH/s |    1/150   |  112.1T  |  0.0032   |  0.0032   |  0.0790   |   0   |   0.332% |  0'C | -42dBm |
> [L/NM] #152 share accepted, 291ms

Cant see any reason but i also  see  a lot of  these lines/errors

> [538772][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():265]: (-32512) SSL - Memory allocation failed
> [538773][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512

> [666394][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1

[log.txt](https://github.com/user-attachments/files/19408170/log.txt)

vanga83, 2025-03-23:
3 out of the 4 I have does this. The 3 that are falling back to 408 KH/s are the ili9341. The only one that's been consistent with staying around 460 KH/s is the st7789.
Also, the 3 ili9341 have this orange rectangle to the left of the BTC price whereas the st7789 does not.

oscar-rr, 2025-03-24:
i got two ili9341 and two st7789. Both have the problem.
The orange icon i can confirm too

vanga83, 2025-03-25:
Just to update, the st7789 that I have has dropped down to 406-408 KH/s and stayed consistent for a day so far. That's all 4 of mine that has dropped down to the 408 KH/s range.

closed 2025-04-13 by oscar-rr