NMminer1024/NMMiner #28: Pool connect

theminer800, 2025-04-02:
It is stuck on connecting to the pool

NMminer1024, 2025-04-03:
> It is stuck on connecting to the pool

Hi sir，Do you mind show me some log details from flash web : https://flash.nmminer.com

[image]

YSR-tech, 2025-04-26:
Had perhaps a similar issue setting up my Lilygo T Dongle S3 on v1.07.01 yesterday. 
It wouldn't resolve pool address properly due to it misreading only part of the pool address and then trying to connect to that randomness. 
As I was mining locally I had to keep adjusting the address input until it finally showed properly on the marketing screen.

This format is what finally worked locally, of course substitute for your needs. Ignore { }

 { ://192.168.1.157:2019 } 

Give it a few to finish the load and transition to the mining screen as it isn't instant.

I presume it would have worked with the local name in similar format but I just went to work with the simple IP address to run the problem down and once I got it working just went with that. 
The :// being the key beginning to the address.

To the Dev: thanks for increasing the efficiency of the Nerdminer, it sure beats only 75KH/s.

Picture of random address chunk it was attempting.

[image]