NMminer1024/NMMiner #30: FOUND A BLOCK ON MY NM-USB-CHAIN-V1 HW v0.5.01

pillo-pillo, 2025-04-12:
Hi there.

I  can't and don´t believe it i did found a block whit my little tiny NM-USB-CHAIN.
Now what will happens.

[image]

[image]

 Thanks.

DualZyclone, 2025-04-12:
disclaimer - I am a crypto newbie and only trying to help

first check your digital wallet 
if not there yet, it could  be waiting to be verified by the mining pool
then check on that mining pool using your digital  wallet address

If you are solo mining BTC and have won a block
Please tell us so we can celebrate you winning the lottery

if not
1) what coin were you mining
2) what Mining Pool are you using and is it SOLO?
3) What type of wallet are you using as mining to crypto exchange wallet risks loosing the block reward

NMminer1024, 2025-04-13:
Hi sir，According to your photo，Firmware Version is v0.x.xx，the block hit is a wrong value and we have fixed this kind of issue in the latest version v1.6.03

pillo-pillo, 2025-04-13:
> 1. what coin were you mining
> 3. what Mining Pool are you using and is it SOLO?
> 4. What type of wallet are you using as mining to crypto exchange wallet risks loosing the block reward

Hi Sean

1. I'm mining **BTC**
2. The pool is **public-pool**
3. BTC wallet form a famous crypto exchanges begins with 3xxxxxx
I will give it a chance 2 month if there's noting in my wallet then I try to upgrade firmware
I remember did'n couldn't upgrade.

Thanks.

But  let you know if's a false positive ore I'm lucky man

pillo-pillo, 2025-04-13:
> Hi sir，According to your photo，Firmware Version is v0.x.xx，the block hit is a wrong value and we have fixed this kind of issue in the latest version v1.6.03

Hi sir.

Well my bliss in a pit, I give't a go for 2 month if no luck with my lucky lottery nm-usb-chain V1 then I upgrade to new version.
I thought I was a lucky man then.

Thanks.

I will let you know.