NMminer1024/NMMiner #46: License Required after Update

IsSasoriDev, 2025-05-07:
after updating and restart, one of my miners was asking for a license, after multiple requests in the console it finally worked, but after a power outage, again, license required, think its a but wich is extremly annoying

NMminer1024, 2025-05-09:
> after updating and restart, one of my miners was asking for a license, after multiple requests in the console it finally worked, but after a power outage, again, license required, think its a but wich is extremly annoying

Hi sir ,
    It's a logic we design: if you purchase the licence from web ever， when you guys upgrade firmware from previous firmware to v1.7.01 or higher，the onchip licence should be inactive ，you’d better wait few seconds until the web send a new licence to your miner, sorry about this .

ussdeveloper, 2025-05-14:
well :) so it's not for free :)

slowride88, 2025-12-29:
Hello,
After my network ssid was changed. 1 of my miners reconnected to the new network. But my second one is now asking for a license. What can I do to fix this issue and get reconnected? I have tried to re-boot and update the firmware. Nothing so far has fixed the problem. Does anyone have a solve for this problem?

slowride88, 2025-12-29:
> https://github.com/NMminer1024/NMMiner/blob/main/README.md says....Note 2: If you lose your license, re-flash your device using the NMMiner Flash Tool<https://flash.nmminer.com/> to receive a new license automatically. Did you update using the flash tool? If you did I would try again..

Im trying to flash the device now. How long does it take to verify a new license?  Its been trying for 15min so far

Regards
Brian Lacey

LsLoki, 2025-12-29:
You need to read the read.me linked above, scroll down for the instructions regarding the licence - I've never encountered this so would be relying entirely on that read.me instruction.  Good Luck