NMminer1024/NMMiner #47: Problem disconnecting equipment in the Public pool

Gustavo-Sindici, 2025-05-08:
Guys, I have a problem with the device disconnecting from the public pool. On the device's display it shows correctly, but when I check the public pool, it doesn't stay in Uptime for more than 1 day.

The image shows devices with different Uptimes, even though they were connected to the network at the same time.

[image]

psionyx2311, 2025-05-09:
Ive never with any device had public-pool give accurate connection times

I think this is more an issue with their web panel rather than the device.

Gustavo-Sindici, 2025-05-09:
Got it, I migrated to taz.mining, let's see how it behaves in the next few days..

thanks for the answers.